gysmo38/mitsubishi2MQTT #195: 2022.12 not working in HA

freddieleeman, 2022-12-12:
After upgrading from 2022.11.1 to 2022.12.0 this morning, I'm unable to control my HVAC from Home Assistant. The set temperature is not visible anymore.

The top HVAC is running 2022.12.0, the bottom 2022.11.1 (again)
[image]

Downgrading to 2022.11.1 solves the issue.

Sprinterfreak, 2022-12-14:
Can You please take a look in Your home-assistant.log for any Tracebacks hinting to some Jinja issues or does it not heat at all?

Have you looked into mqtt traffic for that specific missing value?
Should be at topic mitsubishi2mqtt/+/state
The "temperature" key in that json.

Have you checked the set temp was within the allowed range configured on the ESP?
Check if the desired values appear in topic homeassistant/climate/+/config

The default values are min 16°C, max 31°C. Check on the ESP's webserver if the values are set correct or if any configuration got mangled by the update. Happend to me a few times. Reset to defaults and reconfiguraton fixes this usually.

I'm still on b95ad0a589f516345a2bb39b8b270c85ee129337 which changed that value display in order to be able to display the set temp including min and max. Maybe I try 2022.12.0's latest changes too later.

Sprinterfreak, 2022-12-15:
Just updated one unit to 2022.12.0 and the HA integration works just fine so far.
What I don't like about 2022.12.0 is the chinese language now forcibly active tho.

gysmo38, 2022-12-18:
For me it work fine but with a hight latency. I will not have time to check before christmas break. I will check in January.

sean-leach, 2022-12-19:
I can also confirm that this is happening with the latest 12. update in my Home Assistant install as well. I can control temps if I sign into the device directly, but Home Assistant no longer sees what I currently have set as the temp for the particular unit.

This was working in the 11. release I was using prior.

Sprinterfreak, 2022-12-19:
As i mentioned above. Please provide information.
Since I can not produce this condition with 4 units on 2022.12.0 and HASS 2022.10.2 and  it obviously works for gysmo38 too, we can not further process this issue without external input.

We need at least:
  - commit hash You are on
  - Home Assistant version
  - home-assistant.log
  - set/min/max temp
  - relevant mqtt topics (see above)

jsolarin, 2022-12-27:
It looks like 2022.12 gets an error on the MQTT climate python script trying to parse the float...

Exception in handle_target_temperature_received when handling msg on 'mitsubishi2mqtt/Minisplit_Office/state': '{"roomTemperature":48,"temperature":72,"fan":"AUTO","vane":"AUTO","wideVane":"|","mode":"off","action":"off","compressorFrequency":0}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 572, in handle_target_temperature_received handle_temperature_received( File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 548, in handle_temperature_received payload = render_template(msg, template_name) File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 516, in render_template return template(msg.payload) File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 233, in async_render_with_possible_json_value self._value_template.async_render_with_possible_json_value( File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 589, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1942, in _render_with_context return template.render(**kwargs) File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "<template>", line 1, in top-level template code File "/usr/local/lib/python3.10/site-packages/jinja2/filters.py", line 482, in do_min return _min_or_max(environment, value, min, case_sensitive, attribute) File "/usr/local/lib/python3.10/site-packages/jinja2/filters.py", line 452, in _min_or_max it = iter(value) TypeError: 'float' object is not iterable

I see the following changes to the MQTT config topic

2022.12
  "temp_stat_tpl": "{% if (value_json is defined and value_json.temperature is defined) %}{{ value_json.temperature|float|min(61.00)|max(88.00) }}{% else %}72.00{% endif %}",

2022.11.1
  "temp_stat_tpl": "{% if (value_json is defined and value_json.temperature is defined) %}{% if (value_json.temperature|int > 61.00 and value_json.temperature|int < 88.00) %}{{ value_json.temperature }}{% elif (value_json.temperature|int < 61.00) %}61.00{% elif (value_json.temperature|int > 88.00) %}88.00{% endif %}{% else %}72.00{% endif %}"

I'm on HASS 2022.12.8

jsolarin, 2022-12-27:
Looking at the sources, I can't find where that float is in the master branch... I compiled the binary in the Arduino IDE and did a firmware update. It's good now... I think the precompiled files are using the float and causing HA to error.

Sprinterfreak, 2022-12-27:
User error, not using latest version. Fixed by f000d6e82cefd80347276dc4edf2d6ea201b3902

freddieleeman, 2022-12-27:
I did use the precompiled binary, so that could explain the issue. How is this a user error?

jumojn, 2023-01-14:
Same thing here as freddie has when upgraded home assistant

Sprinterfreak, 2023-01-15:
> Same thing here as freddie has when upgraded home assistant

Binaries are outdated. Please build from source.

freddieleeman, 2023-01-16:
Or... maybe update the binary?

Sprinterfreak, 2023-01-18:
Maybe coffee can help achieve that.

But in general, using the binaries is not recommendable because releases of underlying libraries won't propagate either. You will not find fixed binaries when SwiCago fixes some stuff related to device communication for instance.

JAAS76, 2023-01-21:
Same issue as freddieleeman on 6 units. Downgrading to 2022.11.1 solved the issue.

Home Assistant 2023.1.6
Supervisor 2022.12.1
Operating System 9.4
Frontend 20230110.0 - latest

Shadowpheonix, 2023-03-10:
@Sprinterfreak  said...
> Binaries are outdated. Please build from source.

and...

> But in general, using the binaries is not recommendable because releases of underlying libraries won't propagate either. You will not find fixed binaries when SwiCago fixes some stuff related to device communication for instance.

I have several comments regarding this...
1 - The binaries have the exact same date (December 11, 2022) & version (2022.12.0) as the **released** source code.  Is the **released** source code newer, and if so, how are users supposed to know this?
2 - If the binaries are not intended for use, why were they released?
3 - If we are supposed to use the **un-released** source code, then why are there released versions?

Sprinterfreak, 2023-04-23:
@Shadowpheonix that is because
1. Not every fix is released. The code is kind of rolling. This issue was fixed before filed, before the day of that release without the fix applied.
2. They are. But they are not updated regularly and not contain already fixed bugs.
3. The release step seem to be kind of manual. There is of course work on the way to automate this in the future as far as I see, to ensure integrity and more or less recent state of the binaries.

Basicaly the issue is fixed since Dec 6, 2022 and not made it into the released binaries from Dec 11, 2022. Thats the issue. And a bunch of fixes where made in foreign and this repo since then. Also Unreleased. So if you are experiencing issues, first rule before filing a complaint always is to use latest code. With platformio it's really not that hard.

It's nothing to do here besides waiting for a new release only for those who like binaries.

Sprinterfreak, 2023-05-03:
Also resolved by https://github.com/gysmo38/mitsubishi2MQTT/actions

closed 2023-07-28 by gysmo38