gysmo38/mitsubishi2MQTT #196: Feature request: Flash ESP via Chrome Web Serial API

renne, 2022-12-15:
Hi,

currently OS-specific command-line tools or IDEs are necessary to flash mitsubishi2MQTT on the ESP.
This makes the use of mitsubishi2MQTT impossible for non IT-savy users.

I suggest flashing the ESP via the [Chrome Web Serial API](https://codelabs.developers.google.com/codelabs/web-serial#0).
[Tasmota](https://tasmota.github.io/) for example uses the Chrome Web Serial API [here](https://tasmota.github.io/install/) via the [ESP Web Tools](https://esphome.github.io/esp-web-tools/).

Sprinterfreak, 2022-12-15:
Well, someone not able to not understand basic simple and well documented tasks like flashing an ESP should probably not mess with high voltage electronic equipment internals.

renne, 2022-12-16:
@Sprinterfreak 
Apart from that connecting an ESP is Safety-Extra-Low Voltage not every electrician has to understand how to set up and configure an IDE. A lot of people don't want to litter up their field devices with special tools for specific purposes but use a more general approach.

The propagation of mitsubishi2MQTT will thrive if people can install it without superfluous complexity for testing, learn the benefits and then install an IDE to add features.

Lin4fun, 2023-03-07:
@renne

I promise you that every electricain nowa days is able to program a esp as this is part of the base education of an electrician. I made my excam in 1995 and there it was already a main part to program, solder and create electronic steering.

And even if the connector have just 5 / 3.3 v, you have to take the main power and open up the device which is probihitet in many countries by law for poeple who haven't been schooled to do this.

Actually the way how you try to get this function from a person which is not earning a cent by publishing his nice work is a bit off.

You can buy yourself a working box from melcloud, thats for sure the better solution for you.