gysmo38/mitsubishi2MQTT #197: Chinese language set in config.h

tonylane26, 2022-12-16:
Just started with this and the Webserver pages all come up in Chinese.

The latest version of config.h from a week or so ago includes this line: 
#define MY_LANGUAGE zh.CN // define your language

This seems to conflict/override with the setting of the #ifndef MY_LANGUAGE line in the main .ino file.

I changed it to en-GB but maybe the line shouldn't be in the config.h file at all?

gysmo38, 2022-12-18:
Sorry, this is a mistake from last release push... I compile binary for all language in last release and forgot to comment before push.
When commented, the language is en-GB.

closed 2022-12-18 by gysmo38

tonylane26, 2022-12-18:
> Sorry, this is a mistake from last release push... I compile binary for all language in last release and forgot to comment before push. When commented, the language is en-GB.

I'm still learning so it was useful to debug.

Thanks for all the work you've put into this, as well as to the others whose work you've built on. 
I'm looking forward to trying it "Live" on my new mini-splits, once the builders are finished. 😉