gysmo38/mitsubishi2MQTT #199: support for installer setting/functions

skooj, 2023-01-02:
I have some MSZ-GL06NA that are over-heating a couple of rooms in my house and am hoping to be able to set the installer codes for Thermal Off fan operation.

It looks like SwiCago's HeatPump library is able to handle this:
https://github.com/SwiCago/HeatPump#support-for-installer-settingsfunctions

Skimming the code for this project, nothing is jumping out at me as being able to handle this, but it's possible that I may have missed it. 

Is this something that can be added to Mitsubishi2MQTT?

nicw, 2023-04-10:
skooj, are you talking about the ability to control the fan regardless of whether the temp has been hit or not? I know that MitsuCon (using SwiCago) allowed for that, but I don't see it in this library. Ex: You've hit your target temp, but it's a large room and you want to keep the blower going to spread that air around.

zjklee, 2023-04-10:
@nicw , yes, what you describe is one of those installer settings/functions (at least on my PVA). I'd love this to be added as for some reason I do have this feature turned OFF by default when I'm not using MHK2 ☹️

skooj, 2023-04-10:
@nicw,
as @zjklee said, those are exactly what I am talking about. In the case of the MSZ-GL06NA, (and the like) mini splits, that particular option is always on, and cannot be turned off by a code, but must have a resistor cut on the control board of each indoor unit. However, I still see this being useful for others that have the ability to use the installer codes, hence leaving the request open.

nicw, 2023-04-12:
Gotcha, yes I see your PVA supports it, but not here.  Mine is the jumper kind (and I don't want to cut), so I just have the automation be if the target temp is <x> degrees above the current, run fan at <4>, and then to <auto> when it matches. Within a couple degrees of hysteresis and it works fine. I would like to tune my defrost though, it does it so often, when we don't have ice.

Given that code snippet above is about extracting the codes, you could just take the code and do hp.setValue(code) on the main() loop, as it's creating an hp object from the SwiCago library.

zjklee, 2023-04-12:
Yeah, I may give it a try. However, it may take quite a while as this is going to be my first experience with Arduino 🫣

closed 2024-08-25 by gysmo38