gysmo38/mitsubishi2MQTT #200: Set temperature off by one degree

danepowell, 2023-01-17:
I have several N series mini splits using this library on D1 mini v2 clones. One annoyance I've noticed is that the web interface (and of course MQTT) always report a "set" temperature that is one degree higher than set by the IR remote control. 

For example, if I turn on the unit and set the temperature to 69 degrees F with the IR remote, the web interface says it's set to 70 degrees F.

danepowell, 2023-01-18:
Actually I wonder if this is a rounding error, or a problem converting Celsius to Fahrenheit. When I set the temperature to 68 degrees via the IR remote, it also shows up at 68 via MQTT. But when I change it to 69 via the remote, it jumps to 70 in MQTT.

mbbush, 2023-01-22:
68 F is exactly equal to 20 C, so it makes sense that would get represented accurately. 

And 69 F = 20.5555555 C, which, if rounded to the nearest degree C is 21 C, and 21 C is 69.8 F, which rounds to 70. 

So I think your explanation of the issue as a rounding problem is spot on. It's just a matter of figuring out where in the code the issue happens, and/or how to fix it. My best guess is that something in the code is using the step size (which you have presumably set to 1 degree F) to round in celsius, not just in Fahrenheit. 

To help with debugging, could you make a table of various temperatures (in F) input by the remote, and how they show up in MQTT? Then switch your mitsubishi2mqtt configuration to use Celsius with a step size of 0.5 degrees, and then report how various temperatures (in F) set by the remote are reported (in C) to MQTT. Maybe also try configuring mitsubishi2mqtt to use Fahrenheit with a step size of 0.5 degrees and try the same?

The F/C conversion scale repeats every 9 degrees F / 5 degrees C. So a range of maybe 12 consecutive degrees F (like 66 to 78) would be enough input to see all the possible ways of rounding.

mbbush, 2023-01-22:
Some of the discussion at https://github.com/SwiCago/HeatPump/issues/51 might be relevant here.

danepowell, 2023-01-24:
Oh yeah, C-F conversion is pretty clearly the issue. Based on [the code](https://github.com/gysmo38/mitsubishi2MQTT/blob/3f1f95abf6a10dc75a99807dce0b676535f57847/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1242) and the discussion in https://github.com/SwiCago/HeatPump/issues/51, it sounds like all heat pumps (even in USA) internally track temperature in Celsius, presumably in 0.5 degree increments. The remotes perform some F-C conversion on the fly when working in Fahrenheit mode, and whatever algorithm they use is different than ours. Possible because one of these libraries is dropping the decimal and treating the temps as ints.

I'll try to take a closer look at this later and get the data you asked for.

danepowell, 2023-01-24:
Here is the data for heating mode. Number on the left is IR setpoint, number on the right is value from this library:

72 73
71 72
70 71
69 70
68 68
67 66
66 65

mminehanNZ, 2023-08-30:
Sounds like a similar issues to https://github.com/gysmo38/mitsubishi2MQTT/issues/175
It might be to do with the use of "toInt()" instead of "toFloat()", which causes an issue due to dropping decimal places when rounding.

dsstewa, 2024-07-06:
@danepowell @mbbush Just put in a pull request, this issue was driving me crazy. I have some MA controllers on my units and not having the temps match up was driving me crazy. Take is look if you would like to implement before it gets merged into the main branch dsstewa:correct-fahrenheit-celsius-to-match-remote

closed 2024-09-07 by danepowell