gysmo38/mitsubishi2MQTT #205: power_command_topic is deprecated

freddieleeman, 2023-02-02:
I noticed this warning in my Home Assistant log today. Unfortunately, I could not find the option in my configuration, but since it triggers five times, and I have five Mitsubishi HVACs running mitsubishi2MQTT, I decided to submit the issue here. I haven't had time to dive into the details yet.

home-assistant.log:
`WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'power_command_topic' option is deprecated, please remove it from your configuration`

freddieleeman, 2023-02-02:
I couldn't find the `power_command_topic` string which made me second guess if this was related to this repo, but then I found the abbreviation here: https://github.com/gysmo38/mitsubishi2MQTT/blob/f000d6e82cefd80347276dc4edf2d6ea201b3902/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1546

lesleypersyn, 2023-02-02:
#206

drako92, 2023-02-16:
> PR created to fix this #206

Thanks for the fix, but i necessary have to modify the 3 files and reflash the firmware, or can i directly upgrade the firmware?

closed 2023-07-28 by gysmo38