gysmo38/mitsubishi2MQTT #216: Implement set ventilation to OFF

GhyslainBruno, 2023-04-17:
Hi all, 

I'm currently creating a zoning system from scratch with a PEAD-SM50JA heat pump, controlled by HomeAssistant, some home made physical thermostats, and I'm using this lib (which is really amazing, special thanks to @gysmo38 and to his work, you rock 🙏). 

What I'm actually looking for is a way to disable the ventilation of the HP. 

I'd like to actually have the HP set to `ON`, but not blowing out any air. 

Does anyone know if this is possible ? 

The way I'm doing this right now is by setting the temperature of the HP to the minimum (17°C for mine) when no zone is demanding heating. 

It kinda works, but is not very clean as you can image. 

Anyway, if someone is interested in this topic, the repo can be found [here](https://gitlab.com/ghyslainbruno/zoning). 

Thanks to anyone spending some time reading this. 

GhyslainBruno.

Sprinterfreak, 2023-04-24:
I guess what You are searching for is the "Thermal Off Fan Operation" https://github.com/SwiCago/HeatPump/issues/39#issuecomment-1040924886 Not quiet sure if this is safe to set by now.

There is currently no infrastructure in the code of mitsubishi2mqtt to set such special bytes besides sending raw packets.

Also HASS's climate plugin can't display such special cases. It could be done via additional select's to the auto-discovered device.

Drewlrox, 2023-06-17:
I also would be interested in this feature. I was reading [here](http://enter.mehvac.com.s3.amazonaws.com/DAMRoot/Original/10006/Application%20Note%203048%20ME%20-%20How%20to%20turn%20off%20indoor%20unit%20fan%20when%20set%20point%20is%20met.pdf) that this can only be achieved on my m series unit by using the dip switches on the thermostat interface PAC-US444CN-1. This thermostat adapter/interface connects via the same cn105 connector my ESP8266 is connected to.

I believe this could greatly improve filter life and save some energy. My understanding is that the fan continues to run so the unit can get an accurate temperature reading, but there is an external temperature sensor accessory (2 wire so maybe it could be replaced with a simple thermistor) that could mitigate this.