gysmo38/mitsubishi2MQTT #219: Question - MQTT - unauthorized access

Oq6Jw1Oi4jHR1L, 2023-06-12:
Thank you for all the work for mitsubishi2MQTT! I noticed an issue today when I had to remove all my boards overnight and reinstall them.

The wifi still worked but MQTT refused access until I changed the username.
I'm runnning HAOS and I had specified a hvac user for these boards.

Has anyone else ran into something similar?

sean-leach, 2023-06-13:
I am having the same issue on the latest build of mitsubishi2MQTT, latest HAOS. Two of my units in particular go dark inside Home Assistant. Wifi is still fine, connection is good, and I can access each via the web UI. MQTT just can't see them.

No idea how to fix it. Any ideas?

Oq6Jw1Oi4jHR1L, 2023-06-13:
> I am having the same issue on the latest build of mitsubishi2MQTT, latest HAOS. Two of my units in particular go dark inside Home Assistant. Wifi is still fine, connection is good, and I can access each via the web UI. MQTT just can't see them.
> 
> No idea how to fix it. Any ideas?

I think it might be similar to my issue. If you create a new user to use with MQTTT, does that work?
I had to create a new user and use those credentials as a workaround.

sean-leach, 2023-06-13:
> I think it might be similar to my issue. If you create a new user to use with MQTTT, does that work? I had to create a new user and use those credentials as a workaround.

I will give this a shot, thank you!

sean-leach, 2023-06-17:
Sadly one of my boards still will not connect to HAOS, even after swapping MQTT users out. It's so strange as I can access it via the web UI right into the board, but HAOS can't see it.

sean-leach, 2023-09-09:
Anyone else have any ides for a device that's accessible via IP, but Home Assistant cannot see it?

vperas, 2023-09-09:
> Anyone else have any ides for a device that's accessible via IP, but Home Assistant cannot see it?

This is how I managed to solve it:
- take the device offline.
- remove the device (entity) from HA. 
- download MQTT-Explorer and open it, connect to your MQTT server.
- delete every topic related to the AC (mitsubishi2MQTT and under HA)
[image]
- start the device and it should be found automatically by HA