gysmo38/mitsubishi2MQTT #223: Exception in Home Assistant

heffneil, 2023-07-09:
I upgraded to the latest version and suddenly seeing this in my HA Logs:

Exception in handle_target_temperature_received when handling msg on 'mitsubishi2mqtt/Minisplit/state': '{"roomTemperature":73,"temperature":72,"fan":"AUTO","vane":"5","wideVane":"|","mode":"cool","action":"cooling","compressorFrequency":14}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 681, in handle_target_temperature_received handle_climate_attribute_received( File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 648, in handle_climate_attribute_received payload = render_template(msg, template_name) File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 616, in render_template return template(msg.payload) File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 235, in async_render_with_possible_json_value self._value_template.async_render_with_possible_json_value( File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 716, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2164, in _render_with_context return template.render(**kwargs) File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "<template>", line 1, in top-level template code File "/usr/local/lib/python3.10/site-packages/jinja2/filters.py", line 482, in do_min return _min_or_max(environment, value, min, case_sensitive, attribute) File "/usr/local/lib/python3.10/site-packages/jinja2/filters.py", line 452, in _min_or_max it = iter(value) TypeError: 'float' object is not iterable

mcouture87, 2023-07-09:
The bin files are compiled with float and I had same issue.    I compiled the source code manually and the error went away so, long story short, compile the code and you won't have this problem.

heffneil, 2023-07-09:
Yeah I’m not that skilled. Also if I replace the bin I lose all my settings again? I didn’t notice is there a backup and restore option?

heffneil, 2023-07-09:
I am also not getting temperature:

current_temperature: 73
temperature: null
fan_mode: AUTO
hvac_action: cooling
swing_mode: '5'
friendly_name: Attic Air Conditioning
supported_features: 41

this is causing my automation to be very upset.  I compare the setpoint and the current temp and - well its not showing it.

heffneil, 2023-07-09:
{"roomTemperature":72,"temperature":71,"fan":"AUTO","vane":"5","wideVane":"|","mode":"cool","action":"cooling","compressorFrequency":11}

heffneil, 2023-07-12:
I compiled and believe this is solved.  I have a problem with my AC unit where it stops working and blinks 17 times.  I am unsure if this has ANYTHING to do with it.  My AC guy came and there's nothing in the manual for 17 blinks.  Anyway I have removed it from MQTT in an effort to rule out this unit.

closed 2023-07-28 by gysmo38