gysmo38/mitsubishi2MQTT #225: After updating I can't setup temperature anymore

claustro, 2023-07-14:
Hi all:
I am having problems after updating Wemos D1 to the latest version.
I can't set up the target temperature anymore. I can only switch on or off the machine
I can fix it by deleting the main topic ( AC studio) in mqttexplorer , but the problem arises  another time after some time or if I restart Homeasistant

The topics of the old version are different from the new one.
The old one publish 
settings:
state: 
the new one  publish
setting :
state :
temp :
debug :
availability=

[image]
[image]

robertoamd90, 2023-07-19:
Similar issue, but after delete topic MQTT i'm able tu set tempo moving the point, the temperature is currectlty setted on the clima, but the point and taget temperature return on 22

Similar issue, but after delete topic MQTT i'm able tu set temperature moving the point, the temperature is correctly setted on the machine (and i can verify on the web page of D1), but the point and target temperature return on 22 after 1 second or less.
Thanks

robertoamd90, 2023-07-19:
I have found a workaround.

I have modified the template configuration of auto discovery like in the screen above, and now i can set and read the target temperature on home assistant.

I don't know if the function removed make a worst effect, I'm trying to understand better!

With stock template:
[image]
[image]

with modified template:
[image]
[image]

robertoamd90, 2023-07-19:
> Hi Roberto, can you explain to me better the step you took to solve the problem? I am having the same issue regarding the showed 22 degree after deleting the topic

Sure, in order you have to do in general (if other user need to temporary fix):

1.  install MQTT Explorer and connect to your MQTT broker
2. copy the Value of the topic 'homeassistant/climate/MitsubishiFriendlyName/config0', where MitsubishiFriendlyName is the name you have configured on the ESP Setup -> MQTT -> Friendly name Field
3. paste in a notepad and replace the value for the key 'temp_stat_tpl' from:
`"{% if (value_json is defined and value_json.temperature is defined) %}{{ value_json.temperature|float|min(16.00)|max(31.00) }}{% else %}22.00{% endif %}",` 
to:
`"{% if (value_json is defined and value_json.temperature is defined) %}{{ value_json.temperature }}{% else %}22.00{% endif %}",`
5. go to Home Assistant and delete the Device MitsubishiFriendlyName, Settings -> Devices & Services -> MQTT -> Click on X Devices -> MitsubishiFriendlyName -> click on 3 dots -> Delete
6. on MQTT Explorer publish a new topic equals to step 2, use for the value the modified version of the step 3
7. Now you have again the MitsubishiFriendlyName in MQTT and Home Assistant and the Target Temperature working again  

I don't know if the corrupted configuration is re loaded in any way, with manual or automatic action...

I also discovered reading the source code that bin file on the release i have downloaded and fleshed not corresponding to the source code, because this error was fixed in December! I think a wrong branch is compiled for the bin creation, [The line code incriminated](https://github.com/gysmo38/mitsubishi2MQTT/blob/f000d6e82cefd80347276dc4edf2d6ea201b3902/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1534C26-L1534C26).

I would try to compile the source code from the master branch and test it, but i'm studying how...

claustro, 2023-07-19:
Hi,
I compiled the code using the last release and it seems ok. The only problem I made it in a rush and I compiled the Chinese language! I have to study how to change the language in English.
You can compile it using Arduino software after installing all the needed dependencies
[mitsubishi2mqtt.ino.d1_mini.zip](https://github.com/gysmo38/mitsubishi2MQTT/files/12100240/mitsubishi2mqtt.ino.d1_mini.zip)

robertoamd90, 2023-07-19:
Great work!

After update I had cold sweat because the d1 don't respond, after that i understand the update make a factory reset!

P.S. the version compiled is `Mitsubishi2MQTT 2023.1.0`, the last release `Mitsubishi2MQTT 2022.12.0` (with the bug).

@gysmo38 I would like to thank you for this great project, if you can, please add a new release with last source code!

claustro, 2023-07-28:
No I am not having such problem