gysmo38/mitsubishi2MQTT #229: OTA update resets config

danepowell, 2023-08-03:
I have this library installed on a Wemos D1 mini. I normally compile the sketch myself and push updates wirelessly from the Arduino IDE. I'm pretty sure this normally preserves wifi and MQTT config.

Today I instead downloaded the pre-built binary from GitHub Actions and used the web server OTA update. All of my config was lost when the Wemos D1 mini restarted.

Is this intentional? It's certainly undesirable! 😄 And it causes confusion; like the user in #222, I thought I'd soft-bricked the Wemos until I realized it was in AP setup mode.

lesleypersyn, 2023-08-04:
I did last week OTA updates on 2 D1's with pre-built version. Works fine and settings still exist...

danepowell, 2023-08-05:
Maybe this only happens when switching from Arduino updates to web uploads. Five heat pumps in row lost their config when updating to the latest firmware via the web UI (they were previously compiled and pushed from Arduino IDE). But when I use the web UI to flash the same bin a second time, the config was preserved.

I'll admit I only know enough about Arduinos and the Wemos to get myself into trouble. It makes sense that there's some technical reason why changing upload methods would reset the flash, but I'm not sure exactly what that is.

gysmo38, 2023-08-05:
Maybe the flash layout is not the same between  Arduino and platformio build. It could explain you loose settings.

danepowell, 2023-08-06:
If there's a way to align the flash layout between Arduino IDE and platform.io, I think that would be good (either by documenting the Arduino settings in the README, or adjusting platform.io config). Otherwise you're likely to keep getting issues like #222 where people are confused.

I attempted to fix this in https://github.com/gysmo38/mitsubishi2MQTT/pull/234, but soft-bricked my wemos in the process 😢 It's stuck on the wifi setup page and won't save any settings or reboot. I'll need to extract it from the heat pump.

I think that's my sign to give this up. It's possible #234 is the fix for this but I'm not going to risk further testing. Feel free to keep open or close as you wish.

danepowell, 2023-08-06:
Oh, nevermind! I was able to test it finally and #234 works! We can continue technical discussion in the PR if you wish.

To recover the Wemos, I connected to the setup mode AP directly from my computer (the ad-hoc PSK is the OTA password, which I didn't expect), and even though the web portal was borked I was able to flash the Wemos from the Arduino IDE over the ad-hoc network.

Sawadee2u, 2023-09-09:
> Oh, nevermind! I was able to test it finally and #234 works! We can continue technical discussion in the PR if you wish.
>
> To recover the Wemos, I connected to the setup mode AP directly from my [...]

I am having the same issue with lost config after a success OTA update using the precompiled Bin for WEMOS D1. I am unable to get the Web portal to take the SSID and password for my server. Can you help me understand how to obtain the PSK info? 
Thanks

commit ab0923f7 (2024-01-04): Fix #229: Match Arduino flash layout
commit b9061cd5 (2024-01-04): Merge pull request #234 from danepowell/danepowell-patch-1-1

Fix #229: Match Arduino flash layout
closed 2024-01-04 by gysmo38