gysmo38/mitsubishi2MQTT #233: Upgrade from 0.7.3 fails

cg089, 2023-08-05:
Hi,
I am using Mitsubishi2MQTT 0.7.3 and wanted to do an OTA update.

Unfortunately the update fails with error 5: not enough space.

[image]

I am using D1 mini. Is there any other possibility except removing the D1 from the device and to a serial flash? Disassembling my AC would take forever :(

Thanks!

Sawadee2u, 2023-09-09:
Just tried updating one of my three WEMOS D1 devices. After starting update using the downloaded WEMOS D1 Bin File extracted from the zip file, window reported Update Successful.  But never rebooted. Removed power from A/C system for 2 minutes and reapplied. Module did not reboot and is not taking ip address.

vperas, 2023-09-09:
> Just tried updating one of my three WEMOS D1 devices. After starting update using the downloaded WEMOS D1 Bin File extracted from the zip file, window reported Update Successful. But never rebooted. Removed power from A/C system for 2 minutes and reapplied. Module did not reboot and is not taking ip address. Edit Seems all config data was lost after updating. Can access HVAC_xxxxxx via wifi but using SSID and SSID password cannot connect to server. Not sure how to get PSK info if needed.

Under PSK you just enter your WiFi password - it should work because it works for the rest of us.
If you cant connect, go to your AP or Wifi router and check the logs to see if the Wemos tried to connect to the WiFi.

closed 2025-02-25 by gysmo38