gysmo38/mitsubishi2MQTT #244: HVAC connection stuck

lightoze, 2023-11-16:
Periodically I get a situation when HVAC connection is stuck in disconnected state. Going to web interface and rebooting helps, so probably reconnect logic is not resetting connection state well enough. Perhaps it could be improved so that the connection is restored without the need to manually reboot?

sean-leach, 2023-11-16:
I have the exact same issue, and web UI reboot does fix it

lightoze, 2024-01-31:
I implemented a workaround by adding this in the main `loop()` function:
```diff
         hpConnectionRetries = min(hpConnectionRetries + 1u, HP_MAX_RETRIES);
         hpConnectionTotalRetries++;
         hp.sync();
+        if (hpConnectionTotalRetries > 10 && mqtt_client.state() == MQTT_CONNECTED) {
+            ESP.restart();
+        }
```