gysmo38/mitsubishi2MQTT #245: openHAB MQTT Channel Autodiscovery not working cause name=null

BOFH90, 2023-11-23:
After flashing and connecting to WiFi and setting up the MQTT-connection, openHAB discovered the device, but showed no available channels. After looking through the openhab.log, I discovered that the HA-config could not be adopted, cause the "name" in the config-topic was "null". So I edited the config and changed the name from nullptr to something else, the channels were successfully created. 
 
That's the part, that needs to be adjusted.:
`haConfig["name"]                          = nullptr;`

willyfromtheblock, 2024-02-21:
https://www.openhab.org/addons/bindings/mqtt.homeassistant/

_The HomeAssistant Climate Components is not yet supported._

Are you certain this will work even if this field is set correctly?

BOFH90, 2024-02-22:
Yes, I do. I edited the haConfig to something random for my devices and the OH Autodiscovery worked after that.