gysmo38/mitsubishi2MQTT #249: MSZ-HR25VF - D1 Mini Clone - HA Constantly being unavailable

xtypebox, 2023-12-15:
I have tried the ESPHome version in the HR25VF ac unit and i had the issues with everything being unavailable when i connect to the pump but it works with webversion. I can get the temperature and set the temperature and the mode. So I tried [mitsubishi2MQTT] and i have the same issue. Everything seems connected and if i use the control page in the webpage the ac unit works as expected but in the Home Assistant the AC entity are always being unavailable and i can't get any info. 

Any recomendation?

HappyPlotter, 2023-12-18:
> I have tried the ESPHome version in the HR25VF ac unit and i had the issues with everything being unavailable when i connect to the pump but it works with webversion. I can get the temperature and [...]

Have you checked your MQTT? Maybe the "offline" topic has an retain flag?

zackbennis, 2023-12-23:
@HappyPlotter - I'm in the same boat here, except I seem to be able to control the mini splits via Home Assistant as well. The constant "became unavailable" notices in the history make the data very hard to interpet visually though, so I'd like to get rid of them. Here's the output from the MQTT broker addon, which seems to indicate constant disconnect/reconnect from the endpoints:

```
2023-12-23 10:59:47: Client HVAC_Upstairs_NE has exceeded timeout, disconnecting.
2023-12-23 10:59:47: New connection from xx.xx.xx.54:59721 on port 1883.
2023-12-23 10:59:47: New client connected from xx.xx.xx.54:59721 as HVAC_Upstairs_NE (p2, c1, k15, u'placeholder').
2023-12-23 11:00:23: Client HVAC_basement has exceeded timeout, disconnecting.
2023-12-23 11:00:23: New connection from xx.xx.xx.55:49928 on port 1883.
2023-12-23 11:00:23: New client connected from xx.xx.xx.55:49928 as HVAC_basement (p2, c1, k15, u'placeholder').
2023-12-23 11:02:47: Client HVAC_basement has exceeded timeout, disconnecting.
2023-12-23 11:02:47: New connection from xx.xx.xx.55:51068 on port 1883.
2023-12-23 11:02:47: New client connected from xx.xx.xx.55:51068 as HVAC_basement (p2, c1, k15, u'placeholder').
2023-12-23 11:03:55: New connection from xx.xx.xx.46:57990 on port 1883.
2023-12-23 11:03:55: Client HVAC_1stflrbed already connected, closing old connection.
2023-12-23 11:03:55: New client connected from xx.xx.xx.46:57990 as HVAC_1stflrbed (p2, c1, k15, u'placeholder').
2023-12-23 11:03:59: Client HVAC_Upstairs_NE has exceeded timeout, disconnecting.
2023-12-23 11:03:59: New connection from xx.xx.xx.54:54547 on port 1883.
2023-12-23 11:03:59: New client connected from xx.xx.xx.54:54547 as HVAC_Upstairs_NE (p2, c1, k15, u'placeholder').
2023-12-23 11:04:45: New connection from xx.xx.xx.54:52533 on port 1883.
2023-12-23 11:04:45: Client HVAC_Upstairs_NE already connected, closing old connection.
2023-12-23 11:04:45: New client connected from xx.xx.xx.54:52533 as HVAC_Upstairs_NE (p2, c1, k15, u'placeholder').
2023-12-23 11:08:05: Client HVAC_basement has exceeded timeout, disconnecting.
2023-12-23 11:08:05: New connection from xx.xx.xx.55:51267 on port 1883.
2023-12-23 11:08:05: New client connected from xx.xx.xx.55:51267 as HVAC_basement (p2, c1, k15, u'placeholder').
2023-12-23 11:08:17: Client HVAC_upstairs_NW has exceeded timeout, disconnecting.
2023-12-23 11:08:17: New connection from xx.xx.xx.39:57830 on port 1883.
2023-12-23 11:08:17: New client connected from xx.xx.xx.39:57830 as HVAC_upstairs_NW (p2, c1, k15, u'placeholder').
2023-12-23 11:09:53: Client HVAC_Upstairs_NE has exceeded timeout, disconnecting.
2023-12-23 11:09:53: New connection from xx.xx.xx.54:56222 on port 1883.
```

bfridman, 2024-02-06:
I'm also having the issue.  @xtypebox and @zackbennis I'm finding the WIFI (connecting to it via IP) is also extremely slow to non-responsive.  Are you also having that problem?  I'm wondering if I sever the connection to MQTT (delete the settings) if it will be better.

jman311, 2024-02-06:
I've been messing with this for a while to try to get a stable connection.  Here's what I've found so far (using an ESP-01S):

Originally I had a super spotty connection between the ESP and MQTT.  It would take like 6-8 hours to even first connect.  I have verified the wifi connection is 100% stable and not the issue.

The default first traffic rule in my home LAN was "drop invalid packets".  This is a common network rule.  I found this rule prevents the vast majority of communication between ESP and MQTT server.  Turning off the rule (or adding an exception rule before it) allows them to talk freely.  Instant connection and much less lag on the web interface.  Still got regular drop-outs but this helped tremendously.

I'm not a network guy so can't speak to why the packets are "invalid", but ... my network definitely disapproves.

Secondly.  Was still getting drop-outs at least 2/minute.  Tried various config values, including setting MQTT_KEEPALIVE.  Setting it to a low value or to 0 I still get drop-outs.  Setting it to the max allowable value of 65535, my drop-outs are averaging about 10 per day.  Not perfect, but at least usable now.

TLDR: Communication prevented by dropping invalid network packets.  Setting MQTT_KEEPALIVE 65535 also helps.  Still getting about 10 drop-outs per day.  

J

Friend-Or-F0H, 2024-11-23:
Still seeing dropouts roughly every 60 seconds. Usually comes back within a minute, but I've tried multiple config changes on my ubiquiti network, and nothing seems to change the behavior.