gysmo38/mitsubishi2MQTT #255: Homeassistant mqtt Mitsubishi

bmause, 2024-01-21:
Hallo zusammen, bin doch noch ein ziemlicher Anfänger in alles dingen zu Homeassistant und MQTT.
Ich habe eine Mitsubishi Klimalange, einen ESP8266 geflachst und eingerichtet.
In Homeassistant unter mqtt sehe ich die Mitsubishi Klimaanlage, aber keine Entitäten oder sonst irgendwie was.

kann mir jemand helfen? Was genau müsste ich noch tuen, oder anpassen in der configuration.yaml?

Habe dort noch nichts angepasst.

Danke in voraus

lesleypersyn, 2024-01-21:
@bmause https://chrdavis.github.io/hacking-a-mitsubishi-heat-pump-Part-2/

bmause, 2024-01-21:
> @bmause https://chrdavis.github.io/hacking-a-mitsubishi-heat-pump-Part-2/

danke dir, habe es auch so versucht, aber ich bekommen keine Entitäten oder es sich nicht steuern.
[image]
[image]