gysmo38/mitsubishi2MQTT #259: Connection Problem

user234453, 2024-02-22:
Hi i recently installed the esp in one of my acs and it worked perfectly so i bought more esps. I installed the program without problems conencted it to the ac and nothing my ac is a MSZ-SF35VE i tried multiple esp chips with and without logic shifters i tried swicago original program and got an eror  fc 5a 01 30 02 ca 01 a8  whatever i do nothing works on the status page mqqt connected but hvac status is always disconected i tried it with ESP32 C2 and C3 devkit modules

 Thank you for your help

willembargeman, 2024-08-08:
I have the same issue. I have a MSZ-GC35VA ac. Have you fixed the issue? I see the same error code in the debug packets.
Somewhere I read that the MSZ-GC models are using 9600 instead of 2400bps. But I can't find where to change this in the code.

user234453, 2024-08-08:
Hi, I wasn't able to fix the problem. I tried changing the libraries from esp8266 to the newer esp32 libraries. After that, I tried to comment out the line of code that was monitoring retry attempts that was erroring out. then I was able to flash the esp32 but with no luck. After that, I pulled out my eps8266 from another project, I put it in the ac and it worked perfectly, so in my findings,  the problem myth be with the internal pull-up resistors with the esp32 or simply the program isn't ment for the eps32. I think that is possible that the different models use different comunication speeds but i think that the esp auto finds the correct one because i have different models and the same esp8266 and it works in both units.