gysmo38/mitsubishi2MQTT #262: Pinouts for using this project on a ESP32-WROOM-32?

amit-nz, 2024-03-03:
Hi All,

Apologies if I missed this in the documentation but I've been through it once or twice (inlcuding SwiCago documenation) but ti isn't clear exactly where I need to attach the pins of the CN105 connector.

This is a diagram of the ESP board I'm using. I've already flashed the `mitsubishi2MQTT_ESP32DEV.bin` file to it using esptool. 

[image]

Physically, pins 27 & 28 (top right) as shown on the diagram above are labelled "RX0" and TX0" on my board. Is this where I need to hook up the RX & TX of my unit?

[image]

Paranoid about frying my brand new aircon so just thought I'd clarify with the experts here ;)

vperas, 2024-03-05:
Here is another view from "above":
[image]

Power off the AC (switch off the breaker or unplug the unit) before connecting the ESP to CN105

ESP PIN28 (Tx) to CN105 PIN5 (Rx)
ESP PIN27 (Rx) to CN105 PIN4 (Tx)
ESP PIN15 (5V) to CN105 PIN3 (5V)
ESP PIN14 (GND) to CN105 PIN2 (GND)

PIN1 on CN105 is 12V and you don't connect a wire to it, just leave it blank

After you finish, power up the unit and connect to the ESP over WiFi and click on the "status" button.
If everything went well you will see this:
[image]

vperas, 2024-03-05:
No problem :)
I run 3 x wemos D1mini on my units (I think those ESP8266 are also 3.3V TTL) directly connected to CN105 for more than a year now, all working good. But I get your point; better safe than sorry.

ecodad, 2024-03-19:
What tail are you using?  I just went to hook mine up and the tail I got doesn't fit the cn105 red connector. [image][image]

amit-nz, 2024-03-19:
I’ve got this one, I’ve not yet hooked up the unit but based on photos of the circuit board, I believe it is correct. 

You want the “5p” variant. Then, once you get it you’ll want to use a sharp implement (a needle or similar) to take out all the wires so that the red one goes to 5V, black goes to GND, and I guess whatever colours you use for RX/TX do not matter too much. If you get those the wrong way around it’s just not gonna work, but if you get 5V/Ground mixed up there’s a good chance you’ll roast your ESP board. 

https://www.aliexpress.com/item/1005002904897793.html

ecodad, 2024-03-20:
Or in my case, it looks like I got the plug wiring backwards on the tail and put 12V on RX - thankfully these boards are cheap and I have spares.  I ordered the tails specified above. The D1 Mini is toast.  Will do the wireing correct tonight and then triple check.

amit-nz, 2024-03-20:
You might also consider a Line level reducer. And a 3v3 step down regulator (I use AMS 1117) 

Apparently you’re not supposed to run any ESP on 5 volts to the pins, or GPIO but this is what the aircon provides. 

In the past I’ve run all sorts if ESPs over 5 volts, some for years on end and not had any issues but most of them were just for things like lighting etc. 

I don’t mind if those blowout, I can easily get to them. But this one is going to be in my ceiling on the air handler (ducted system) so I’m motivated to do it properly!

amit-nz, 2024-03-28:
@vperas it works. 
[image]

I ended up  putting the 5V from the CN105 into a AMS1117 LDO board to step voltage down to 3.3V; then I powered my ESP32 straight via the 3V3 pin. I also used a logic level shifter so the esp never sees any voltages over 3.3v.

closed 2024-03-28 by amit-nz

vperas, 2024-03-29:
Great news, glad you made it work! :)