gysmo38/mitsubishi2MQTT #263: Temperature resets on network interruption

evincent207, 2024-04-10:
Hello, I've recently changed my router and I've started seeing that anytime I bounce the router, my heatpumps reconnect with a temperature of 61F. This has been happening every time I make any changes to the router and bounce. In Home Assistant my schedules remain on but I have to toggle to get temperatures back to the correct value. I don't know Arduino or MQTT well so I'm not sure where the problem might be stemming from. 
If I turn on MQTT debug logging I can see the state values change around when the network briefly goes offline:

                {
                  "payload": "{\"roomTemperature\":73,\"temperature\":68,\"fan\":\"AUTO\",\"vane\":\"AUTO\",\"wideVane\":\"SWING\",\"mode\":\"heat\",\"action\":\"heating\",\"compressorFrequency\":0}",
                  "qos": 0,
                  "retain": 0,
                  "time": "2024-04-08T15:55:33.426769+00:00",
                  "topic": "mitsubishi2mqtt/masterBedroomHP/state"
                },
                {
                  "payload": "{\"roomTemperature\":73,\"temperature\":61,\"fan\":\"AUTO\",\"vane\":\"AUTO\",\"wideVane\":\"SWING\",\"mode\":\"heat\",\"action\":\"heating\",\"compressorFrequency\":0}",
                  "qos": 0,
                  "retain": 0,
                  "time": "2024-04-08T15:55:33.432112+00:00",
                  "topic": "mitsubishi2mqtt/masterBedroomHP/state"
                },

I tested turning off MQTT, and I found that the Arduino GUI had stable target temperatures, so that has me thinking that even with network issues, it is an MQTT/HA issue and maybe not Arduino. That said, if this isn't a symptom of bad packets or something, is there a way I can change the Arduino to protect from this behavior by reading the current temperature from the HP before resetting the temperature? Or changing that reset value to something reasonable? 

I am using Adafruit Feather HUZZAH ESP8266 boards, with my MSZ-FS09NA/MSZ-FS12NA/MSZ-FS15NA mini splits, mosquitto MQTT, home assistant. Asus mesh with 2.4GHZ dedicated IOT SSID, 802.11ax disabled, other recommended settings disabled, 11b disabled, roaming disabled.

evincent207, 2024-04-17:
I've found a good enough solution for my issue.

Looking at debugging with my MQTT I found frequent disconnects and reconnects, but those turned out to be inconsequential. I also found "Client X disconnected due to malformed packet" showing up occasionally. These events appeared to be tied to my temperature resets.

I am running Mosquitto broker in a docker container, and I found a reference to putting my Mosquitto container in host network mode. I was comfortable doing so since I run HA with host network mode, and I haven't had issues since. I was not able to track down the actual reason for the malformed packets, or why the temperature would get set back to 61 after those events.

Reference: https://stackoverflow.com/questions/75534407/mqtt-malformed-packet

As another user stated, there probably is some other issue I just don't understand. I may have needed to have a bridge for both containers. I did see ephemeral source ports changing with the sockets for my MQTT connections, so it may be related to that.

TheRykin, 2024-04-17:
I tried to power cycle my HA, my wifi, and the minisplits themselves and it made little difference.

Looking at your post, I decided to try debugging everything myself, and now the mini splits seem to be holding their set temperature. 
All I had to do was set- MQTT topic debug packets: ON

I don't know enough about how MQTT works to give a rational as to why this works, but I hope that someone smarter than me sees this comment and it helps them understand the issue.

closed 2025-06-02 by evincent207