gysmo38/mitsubishi2MQTT #265: MQTT set power (OFF) and set mode (ON) seem not to work

p-ranit, 2024-04-28:
Hi,
I uploaded the firmware to wemos D1 mini and successfully connected to my HVAC. From the web page everything works great (on/off, setting mode, temperature, etc.).
Using mqtt (mosquitto_pub -d -u xxx -P yyyy -t mtsb2mqttsala/Split_Sala/power/set -m OFF) I can't turn off the HVAC system. Likewise (mtsb2mqttsala/Split_Sala/mode/set -m ON) I can't turn on the HVAC. Conversely, if I send the command mtsb2mqttsala/Split_Sala/mode/set -m HEAT the HVAC is powered and starts heating.
Where am I wrong?
I have another doubt: the guide states that I can use topic/power/set OFF but not topic/power/set ON. Did I get it right?

lapmat, 2024-06-02:
Hi, same problem for me :-(

mesistostefre, 2024-07-15:
Same here: on/off not working. 
Also maybe related: sometimes (8 out of 10) when starting cooling using MQTT, the hvac turns itself off after 3s. It never turns itself off when I use the remote.

funkzy, 2024-08-17:
Same problem - try $topic/mode/set OFF (use 'mode' instead of 'power')

FistandantilusDD, 2025-05-30:
Hi,

I am facing the same issue. I have integrated the AC in Home Assistant. The device is autodiscovered there and can be fully controlled via Home Assistant.
I have additionally exposed the device via matterbridge to Alexa. Alexa is not able to switch it on or off vith the voice commands. I can set the mode to heating or cooling or also change the temperatures. Setting the mode to off is working via App too.
But as written for the simple on/off I am getting this feature is not supported.
Would be great if you further could analyze the issue.

thx
F.

doanerock, 2025-05-31:
I only use [mitsubishi2MQTT](https://github.com/gysmo38/mitsubishi2MQTT) via home assistant and I have had no issues turning the units on/off for heat and AC.  I do like using home assistant since all the units are controlled from one location.  I only used the web page for the initial set up.

dzungpv, 2025-05-31:
It is not working because code not handle it for now, but my fork support it, you can try to apply it to this project or try my fork.
The difference mention here: https://github.com/dzungpv/mitsubishi2MQTT/issues/53#issuecomment-2924323993