gysmo38/mitsubishi2MQTT #266: mqtt server doesn't work with DNS names, only IPs

laviddichterman, 2024-05-28:
Hi! First of all, this project is great and thanks for doing it.

My issue: I can't use my DNS name for my mqtt server and instead need to enter the IP address. Not a huge deal, but the page should indicate that the IP address is required.

Additionally, in the case that the DNS name is used, the ESP board seems to get very sluggish. I assume this is from constantly retrying the MQTT connection?

Easy fix is just to indicate (or even validate) that the mqtt server input is an IP address.

Seems like this could be related to this issue in the underlying PubSub client library: https://github.com/knolleary/pubsubclient/issues/375

floatplane, 2024-05-28:
> My issue: I can't use my DNS name for my mqtt server and instead need to enter the IP address. Not a huge deal, but the page should indicate that the IP address is required.

I personally haven't observed this particular issue - I'm using a CNAME in my [fork](floatplane/MitsuQTT) of this project and it works fine. That said, I've diverged a bit so YMMV 🤷. But I don't remember this ever being an issue.

[image]

> Additionally, in the case that the DNS name is used, the ESP board seems to get very sluggish. I assume this is from constantly retrying the MQTT connection?

How are you determining sluggishness? There's some inherent slowness just from the way the HTTP responses are built up in this project (for example, sending all the JS with every response).

> Seems like this could be related to this issue in the underlying PubSub client library: https://github.com/knolleary/pubsubclient/issues/375

I don't think that bug can be an issue for this project, because this project passes a string pointer that's kept in scope by a global (`mqtt_server`):

https://github.com/gysmo38/mitsubishi2MQTT/blob/b9061cd513a9e65b662ec72188e895f66eb4de84/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L459

laviddichterman, 2024-05-28:
Your fork looks cool! I'll check it out.

> How are you determining sluggishness? There's some inherent slowness just from the way the HTTP responses are built up in this project (for example, sending all the JS with every response).

I mean, the HTTP response often times out. Like, something is occupying those ESP cycles rather aggressively. Under normal conditions the web UI loads in under 1 second.
> 
> > Seems like this could be related to this issue in the underlying PubSub client library: [knolleary/pubsubclient#375](https://github.com/knolleary/pubsubclient/issues/375)
> 
> I don't think that bug can be an issue for this project, because this project passes a string pointer that's kept in scope by a global (`mqtt_server`):

Well, then please disregard that diagnosis.

Specifically I have a pihole setup that assigns my mqtt server the inventive name of 'mqtt.local'
Making that change to the config on this page causes the connection to not work AND the web UI to become unresponsive.
[image]

floatplane, 2024-05-29:
@laviddichterman i have a similar setup, using pi-hole to provide local DNS. 

One thought is that using `.local `sometimes triggers the use of Bonjour/mDNS as opposed to vanilla DNS, though I have no idea what the networking stack does here. Maybe that’s causing issues?