gysmo38/mitsubishi2MQTT #273: Compile error- no matching function for call to 'min(unsigned int, const uint32_t&)'

agrieco, 2024-07-07:
I'm fairly new to arduino and am trying to get this code compiled and loaded on a NodeMCU (ESP32-S).  I'm getting this error and can't seem to figure out what is going on.  Any help would be appreciated

thanks

/Users/user/code/kumo/mitsubishi2MQTT/src/mitsubishi2mqtt/mitsubishi2mqtt.ino: In function 'void loop()':
/Users/user/code/kumo/mitsubishi2MQTT/src/mitsubishi2mqtt/mitsubishi2mqtt.ino:1840:34: error: no matching function for call to 'min(unsigned int, const uint32_t&)'
 1840 |         hpConnectionRetries = min(hpConnectionRetries + 1u, HP_MAX_RETRIES);
      |                               ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/algorithm:61,
                 from /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/Arduino.h:188,
                 from /private/var/folders/dx/ktt4_91n0nz5749mxmdkv05h0000gn/T/arduino/sketches/134041A32792B79C81D5A1880EF0CD7E/sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algo.h:5736:5: note: candidate: 'template<class _Tp, class _Compare> constexpr _Tp std::min(initializer_list<_Tp>, _Compare)'
 5736 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algo.h:5736:5: note:   template argument deduction/substitution failed:
/Users/user/code/kumo/mitsubishi2MQTT/src/mitsubishi2mqtt/mitsubishi2mqtt.ino:1840:34: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned int'
 1840 |         hpConnectionRetries = min(hpConnectionRetries + 1u, HP_MAX_RETRIES);
      |                               ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algo.h:5726:5: note: candidate: 'template<class _Tp> constexpr _Tp std::min(initializer_list<_Tp>)'
 5726 |     min(initializer_list<_Tp> __l)
      |     ^~~
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algo.h:5726:5: note:   template argument deduction/substitution failed:
/Users/user/code/kumo/mitsubishi2MQTT/src/mitsubishi2mqtt/mitsubishi2mqtt.ino:1840:34: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned int'
 1840 |         hpConnectionRetries = min(hpConnectionRetries + 1u, HP_MAX_RETRIES);
      |                               ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/specfun.h:45,
                 from /Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/cmath:1935,
                 from /Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/math.h:36,
                 from /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal.h:30,
                 from /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/Arduino.h:36:
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algobase.h:278:5: note: candidate: 'template<class _Tp, class _Compare> constexpr const _Tp& std::min(const _Tp&, const _Tp&, _Compare)'
  278 |     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
      |     ^~~
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algobase.h:278:5: note:   template argument deduction/substitution failed:
/Users/user/code/kumo/mitsubishi2MQTT/src/mitsubishi2mqtt/mitsubishi2mqtt.ino:1840:34: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
 1840 |         hpConnectionRetries = min(hpConnectionRetries + 1u, HP_MAX_RETRIES);
      |                               ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algobase.h:230:5: note: candidate: 'template<class _Tp> constexpr const _Tp& std::min(const _Tp&, const _Tp&)'
  230 |     min(const _Tp& __a, const _Tp& __b)
      |     ^~~
/Users/user/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_algobase.h:230:5: note:   template argument deduction/substitution failed:
/Users/user/code/kumo/mitsubishi2MQTT/src/mitsubishi2mqtt/mitsubishi2mqtt.ino:1840:34: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
 1840 |         hpConnectionRetries = min(hpConnectionRetries + 1u, HP_MAX_RETRIES);
      |                               ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: no matching function for call to 'min(unsigned int, const uint32_t&)'

Devil1942, 2024-07-20:
Hi, i have the same problem :(

Devil1942, 2024-07-21:
hio, i have downgrade my board library for the esp32 from v3.0 to v2.0.17 and now it runs.

hoenow, 2024-08-23:
Thank you for your solution.
I had exactly the same error message when compiling and was able to fix it by downgrading the library for the esp32 from v3.0 to v2.0.17.