gysmo38/mitsubishi2MQTT #281: remote_temp doesn't seem to work anymore

Sprinterfreak, 2024-12-15:
I have a setup for about 2 Years now and used remote_temp quiet extensively due to up-high wall mounted units getting stuck in building their own hot bubble of air around them and feeling satisfied with that. The room itself stays cold. 
I have therefore put temperature sensors low and fed the float value via mistubishi2mqtt/HVAC/remote_temp/set to the units in 60s intervals.
This worked at some point very well.
Today I noticed (because it's getting cold again) on a 2023.1 firmware, that the units don't seem to use the remote_temp anymore. (Measured by cold feet)
After an update to 2024.8 and seeing in the code that an remote_temp required interval of just 30s was introduced at some point, I changed my sensors to report every 10 seconds and later every 25 seconds (both with the same result).
Now the outdoor unit is just ramping up and down and not heating continously anymore.

Things that come to mind. A poll cycle of all registers usually takes longer than 30 seconds on the indoor units. How should hammering the mitsubishi2mqtt gateway with remote_temp values should ever work?

Other than that, The whole residual AC units are not precision climate mashines and take sometimes 10 Minutes to ramp. Why should we discard remote_temp values after just 30 seconds (shorter than mitsubishi2mqtt can process it)?

Either way, seems entirely broken and also currently breaks the basic operation on the entire hvac system.

remote_temp is important, guys.

notentirelysure85, 2024-12-16:
The current version of the code has this line in config.h :

CHECK_REMOTE_TEMP_INTERVAL_MS = 300000; //5 minutes

My understanding is that remote_temp needs to be sent at least this often, otherwise mitsubishi2mqtt will time out and revert to using the unit's default temperature sensor.

I'm running 2024.8.1 and I don't have your issue. I make extensive use of remote_temp for three units in my house. I've set CHECK_REMOTE_TEMP_INTERVAL to 15 minutes before compiling, as my automation in Home Assistant runs either every 10 minutes, or when there is a sensor update. This has worked fine for quite some time and continues to do so.

When did it stop working for you? I'd suggest installing MQTT Explorer, and watching the remote_temp to see if it is updating correctly.

Also the MQTT path you specified has a typo - mistubishi2mqtt/HVAC/remote_temp/set - Not sure if you've copied it from your config or typed it manually, but still worth checking.

floatplane, 2024-12-16:
FWIW I had serious problems with remote temp and eventually ended up [forking](https://github.com/floatplane/MitsuQTT) to get the behavior I wanted. Specific issues I observed:

1. The timeout check for remote temp isn't rollover-proof and the 32-bit ms clock rolls over in ~50 days. This can cause intermittent issues. You can reboot your ESP every ~30 days to avoid issues (post `reboot` to the `system/set` topic). Or you can wrap your time intervals in something that handles rollovers and does the interval math correctly.
2. Reverting to the internal temperature sensor for me was disastrous - I'm controlling an air handler which lives in an unheated basement space, where the temperature sensor always reads a very low temp. If the remote temp is no longer valid, the controller starts reading a really low temperature and blasts the living area with heat. For my scenario, I'd rather just turn the unit off than revert to the internal sensor. 

YMMV of course!

Sprinterfreak, 2024-12-17:
@notentirelysure85 
Yeah, the naming for these variables irritated me. Seems like intervals between 30 sec and 300 sec are valid. But that part of the code seems to be also broken in a different way (See below).
Can't say when the issue started because the symptoms, at least in my case, are rarely visible. It's just obvious if the unit manages to build a temperature barrier mid room and maintains it well while the remote sensor is clearly multiple degrees below set point. Doesn't always happen, but if, only a towel helps stirring up the air so the unit gets out of its warm comfort zone and starts back up again.
Eagle eye, good catch with the typo. I have verified this and it's unfortunately indeed just in the issue.

@floatplane
May very likely at least contributing to this case, yes. And also underlines the importance of remote temperature sensing. Ideally the default failover behavior schould be configurable via UI. At least ['Revert to internal', 'Unit power off']. To get a little more transparency about the current state, we should add a sensor for homeassistant which reflects the current temp source ['internal', 'external'] in the periodic state updates.

After a reboot, the units seem to behave a little different indeed, although multiple things have been changed.
They where:
 * Update interval raised from 30 sec, 60 sec to now 180 sec
 * Float precision reduced from .01 to .1
 * Units rebooted

Kinda hard to say what really made a difference. Unfortunately the units do not seem to react instantly to changes and there is no direct feedback what the unit sees and accepted. This makes debugging hard.

What changes did I observe?
 * Fans tend to stop now if units are idling
 * looks like the whole system runs more continously on lower compressor speeds instead of cycling every half an hour

Don't know if it's related, but hope it is. Any thoughts or similar experiences anyone?