gysmo38/mitsubishi2MQTT #284: Issue/note about migrating from Honewell MHK1 thermostats

Mortimer452, 2025-01-27:
Not necessarily an "issue" but worth noting that there is a setting that must be changed from the Honeywell MHK1 controller before switching over to the ESP32/Mitsubishi2MQTT device.

In the [MHK1 installation manual](https://www.mylinkdrive.com/viewPdf?srcUrl=http://s3.amazonaws.com/enter.mehvac.com/DAMRoot/Original/10004\Mr_Slim_Series_MHK1_Central_Remote_Controller_Install_12-10.pdf) page 7 lists the installer setup tables.  Setting# 173 is set to 1 (Sense at Remote Controller) by default which tells the unit to read the current room temperature from the controller rather than the indoor unit itself.

If this setting is not changed to 0 (Sense at Indoor Unit), after installing the ESP32 board, the air handler is forever "stuck" thinking the room temperature is whatever value was last read from the MHK1 thermostat.  For example, if your MHK1 set temp was 68 and the room temp was 64 when you replaced the MHK1, the ESP32 thinks the room temp is 64 forever, continuously blasting the heat trying to get it up to the set temp of 68 which it never reads.

The workaround is, *before* removing the MHK1 and RedLink adapter:

- Press and hold the FAN and TEMP UP buttons for ~5 seconds to get into the installer settings.
- Press "Next" until setting 173 appears
- Press the DOWN TEMP button to ensure this setting is set to 0
- Press Done to save the setting

floatplane, 2025-01-27:
You wouldn’t want to do this in all cases though. My air handler is in an unconditioned crawl space and I never want to use the temperature sensor there. Instead, I pull temperature from the room and push it to the handler with `remote_temp` - basically replacing what the MHK1 used to do.

Mortimer452, 2025-01-27:
Yes very true, not necessarily an issue but worth noting for those moving from the MHK1 to this solution.  If you leave this set to "Sense at remote" and never setup automations/node-red to periodically supply the room temp, the unit cannot read the actual room temp.  As far as I can see there's no way to switch it back to "Sense at unit" without re-connecting the old MHK1.  Maybe worth adding a section in the Setup section to turn on/off this setting?