gysmo38/mitsubishi2MQTT #289: Wifi disconnecting

jbserious, 2025-05-13:
Hi all,

I need some help with this one.
I have flashed this repo onto a Weimos D1 mini V1 but it seems to be disconnecting the wifi since a while.
It has been working just fine work a couple of weeks.
Currently at home i am using a Unifi 6 lite and it is connected to 2.4GHZ.

Can you please guide me on how i can troubleshoot this?
Is there a log i can look trough on the Weimos itself?
In the Unifi controller it is only showing disconnected but nothing more than that.

Apologies in advance for the lack of info.

doanerock, 2025-05-13:
I have a similar setup but I have not had any such disconnection issues.  I am assuming the signal strength in that area is normally pretty high.

jbserious, 2025-05-13:
Yes correct the AP is less than 3 meters with a thin wall in between.
I am thinking of re-flashing it tomorrow just to check.

I have another Weimos D1 mini running down stairs as well but that is working just fine.

jbserious, 2025-05-14:
I re-flashed the D1 mini yesterday but the issue remains.
I think i flashed it correctly but all my MQTT and Wifi information were still present after the re-flashing.
Unfortunately i am not that skilled with Arduino IDE but i just open the INO file and flashed it to the board.
Is that the correct way?

jbserious, 2025-05-14:
> I'm having the same issue. Unifi nanoHD using a dedicated 2.4Ghz 'iot' network. Haven't had trouble or drops with other devices. Looks like it was online for a couple hours before disconnecting.

Do you know which Wemos D1 mini version you have.
I have both a V1 and a V3 where the V3 is way way more stable.

Now i have the V1 mini connected to a Xiaomi 65W powerbank to see if it is the AC unit or the Wemos. but it seems to be the Wemos unit.
I have order another V3 via Aliexpress to see if that will fix the issue.

mgillman, 2025-05-14:
Mine is a V3 clone from Amazon. I have a few more, so I'll have to flash another to see if its more stable.

mgillman, 2025-05-19:
Determined how to fix the issue, but still haven't found the root cause.

I now have 3 of these devices setup and properly connected to their split units. They're working just fine communicating w/ Home Assistant via MQTT. No dropouts, very responsive. However one by one I lost the ability to reach the web UI directly. Never lost the ability to operate them via MQTT, so I knew they're running OK.

Determined that I could ping them from other devices on my network, and curl could retrieve the web UI.
But I couldn't ping them from my Mac (15.4.1)

The issue was that my Mac cached an 'incomplete' routing. Like:
`me@machine ~ % arp -a | grep 192.168.2.123
? (192.168.2.123) at (incomplete) on en0 ifscope [ethernet]`

So I deleted the record and created it manually (w/ wemos MAC address):
`me@machine ~ % sudo arp -d 192.168.2.123
me@machine ~ % sudo arp -s 192.168.2.123 40:91:41:2c:f9:82`

I find that if I delete it and don't manually provide the MAC address, it again will fail to ping 192.168.2.123. Not sure why my laptop is having trouble finding these devices on its own. Is the Wemos not replying to my ARP requests?

jbserious, 2025-06-02:
I have since bought another V3 and made the change to clip one of the legs of the board as mentioned here: https://community.home-assistant.io/t/mitsubishi-ac-with-wemos-d1-mini-pro/107007/147 and it has worked perfectly ever since.
No more WiFi disconnect issues.
I suspect mine was just a bad clone board nothing more.