gysmo38/mitsubishi2MQTT #290: Upgrade 0.4.7.2 to last version

alexmc1510, 2025-05-18:
Hello,

I am redoing all my Hassio configuration and I see that for mitsubishi2MQTT a huge improvement has been done since my first installation (0.4.7.2). My question is, is there a way to upgrade my version to the last one without the need of disassembling the module from the AC? (OTA)?

I don't see the button Firmware update in my home screen.

Thanks in advance,

doanerock, 2025-05-20:
Create bin file by compiling code.  For example with  with VScode and platformio https://community.platformio.org/t/how-to-make-bin-file-using-platformio-ide/8491/4 
Go to mitsubishi2MQTT website and click "Firmware Upgrade" and upload bin file, done.

alexmc1510, 2025-05-20:
Thanks for the explanation...but this button does not exists in my version.

lesleypersyn, 2025-05-20:
Adjust platformio.ini and upload. that should work. OTA was already integrated. (not via page/button)
https://docs.platformio.org/en/latest/platforms/espressif8266.html#over-the-air-ota-update

alexmc1510, 2025-05-21:
Thanks for the explanation. I followed your comment and worked perfectly. I didn't know that platformio had this capability.
Thanks again!